Bastelschlumpf/M5PaperWeather #1: Not refreshing

7samurai, 2021-05-20:
Hi,
great script, I got it to work. What does not work, is that the data update themselves. I am not so good in arduino, what could be the reason?

Bastelschlumpf, 2021-05-24:
Hello,
Can you explain a little more what you mean?
Have you registered with openweathermap and are weather data displayed?
The system is set to go into deep sleep mode after displaying for one hour to save power and then fetch and display the weather data again.
ShutdownEPD(60 * 60); // every 1 hour
You can test this by setting it to 10 seconds.
ShutdownEPD(10);

7samurai, 2021-05-25:
Hi, 
i get all the data incl. the time from openweathermap , but, if is set ShutdownEPD(10), there come no new date after an period of time:

M5EPD initializing...OK
Connecting to ****
.....WiFi connected at: 192.168.1.52
batteryVolt: 4.28
batteryCapacity: 93
Epochtime: 1621934688
DateTime: 25.05.2021 09:24:48
Latitude: 53.87
Longitude: 10.69
WifiRSSI: -38
BatteryVolt: 4.28
BatteryCapacity: 93
Sht30Temperatur: 26
Sht30Humidity: 37
MoonRise: 25.05.2021 18:18:07
MoonSet: 25.05.2021 02:35:24
Sunrise: 25.05.2021 05:00:27
Sunset: 25.05.2021 21:28:16
Winddir: 220.00
Windspeed: 2.57
WeatherDisplay::Show
Stop WiFi
Shutdown

7samurai, 2021-05-27:
I looks like, the display never wake up from its shutdown.

damianharouff, 2022-04-20:
Is your M5paper plugged in to USB? If an M5paper is plugged in, it will never shut down. This info can be found on the M5paper's forums.

Mine would also not update the weather data, until I unplugged it from USB.

Hamburgpriat, 2022-08-11:
Hi,
i have the same problem. strangely it updates the data when the cable is CONNECTED.
As soon as i unplug the cable, the M5 will not update. I can't get it solved unfortunately.

tml89, 2022-08-19:
If the m5paper ist plugged, Shutdown will not fired. Add delay(xxxx) after ShutdownEPD() and call a refresh method

Hamburgpriat, 2022-08-19:
> If the m5paper ist plugged, Shutdown will not fired. Add delay(xxxx) after ShutdownEPD() and call a refresh method

the funny thing is that the shutdown works when it is plugged in. and when it is not plugged in, it does not work. so exactly the other way around than the manufacturers say.

drewsed, 2022-12-04:
I think you are using another fork, where not M5.shutdown() is being used but the esp is put to sleep. But then the battery is cut before and it never wakes up as the RTC ia not set for wakeup. Just put the first command there and it's fine.

Hamburgpriat, 2022-12-09:
> I think you are using another fork, where not M5.shutdown() is being used but the esp is put to sleep. But then the battery is cut before and it never wakes up as the RTC ia not set for wakeup. Just put the first command there and it's fine.

Hi. With the original and newest code from Bastelschlumpf it works now. Thank you very much. Then I rebuild my code from his project and the forks myself, as I want it. Thank you.