Bastelschlumpf/M5PaperWeather #6: Arduino compile error

dronix27, 2021-11-27:
In file included from D:\Arduino\M5PaperWeather\weather\Data.h:24,
                 from D:\Arduino\M5PaperWeather\weather\weather.ino:30:
D:\Arduino\M5PaperWeather\weather\Weather.h: In member function 'bool Weather::Fill(const JsonObject&)':
D:\Arduino\M5PaperWeather\weather\Weather.h:127:75: warning: 'typename ArduinoJson6185_91::enable_if<ArduinoJson6185_91::is_same<T, char*>::value, const char*>::type ArduinoJson6185_91::MemberProxy<TParent, TStringRef>::as() const [with T = char*; TObject = ArduinoJson6185_91::ElementProxy<ArduinoJson6185_91::MemberProxy<ArduinoJson6185_91::MemberProxy<ArduinoJson6185_91::ObjectRef, const char*>, const char*> >; TStringRef = const char*; typename ArduinoJson6185_91::enable_if<ArduinoJson6185_91::is_same<T, char*>::value, const char*>::type = const char*]' is deprecated: Replace as<char*>() with as<const char*>() [-Wdeprecated-declarations]
       hourlyMain[0]    = root["current"]["weather"][0]["main"].as<char *>();
                                                                           ^
In file included from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:9,
                 from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/BasicJsonDocument.hpp:7,
                 from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/DynamicJsonDocument.hpp:7,
                 from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:21,
                 from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from D:\Arduino\M5PaperWeather\weather\Weather.h:25,
                 from D:\Arduino\M5PaperWeather\weather\Data.h:24,
                 from D:\Arduino\M5PaperWeather\weather\weather.ino:30:
D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Object/MemberProxy.hpp:79:7: note: declared here
       as() const {
       ^~
In file included from D:\Arduino\M5PaperWeather\weather\Data.h:24,
                 from D:\Arduino\M5PaperWeather\weather\weather.ino:30:
D:\Arduino\M5PaperWeather\weather\Weather.h:128:75: warning: 'typename ArduinoJson6185_91::enable_if<ArduinoJson6185_91::is_same<T, char*>::value, const char*>::type ArduinoJson6185_91::MemberProxy<TParent, TStringRef>::as() const [with T = char*; TObject = ArduinoJson6185_91::ElementProxy<ArduinoJson6185_91::MemberProxy<ArduinoJson6185_91::MemberProxy<ArduinoJson6185_91::ObjectRef, const char*>, const char*> >; TStringRef = const char*; typename ArduinoJson6185_91::enable_if<ArduinoJson6185_91::is_same<T, char*>::value, const char*>::type = const char*]' is deprecated: Replace as<char*>() with as<const char*>() [-Wdeprecated-declarations]
       hourlyIcon[0]    = root["current"]["weather"][0]["icon"].as<char *>();
                                                                           ^
In file included from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:9,
                 from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/BasicJsonDocument.hpp:7,
                 from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/DynamicJsonDocument.hpp:7,
                 from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:21,
                 from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from D:\Arduino\M5PaperWeather\weather\Weather.h:25,
                 from D:\Arduino\M5PaperWeather\weather\Data.h:24,
                 from D:\Arduino\M5PaperWeather\weather\weather.ino:30:
D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Object/MemberProxy.hpp:79:7: note: declared here
       as() const {
       ^~
In file included from D:\Arduino\M5PaperWeather\weather\Data.h:24,
                 from D:\Arduino\M5PaperWeather\weather\weather.ino:30:
D:\Arduino\M5PaperWeather\weather\Weather.h:133:84: warning: 'typename ArduinoJson6185_91::enable_if<ArduinoJson6185_91::is_same<T, char*>::value, const char*>::type ArduinoJson6185_91::MemberProxy<TParent, TStringRef>::as() const [with T = char*; TObject = ArduinoJson6185_91::ElementProxy<ArduinoJson6185_91::MemberProxy<ArduinoJson6185_91::ElementProxy<ArduinoJson6185_91::ArrayRef>, const char*> >; TStringRef = const char*; typename ArduinoJson6185_91::enable_if<ArduinoJson6185_91::is_same<T, char*>::value, const char*>::type = const char*]' is deprecated: Replace as<char*>() with as<const char*>() [-Wdeprecated-declarations]
             hourlyMain[i]    = hourly_list[i - 1]["weather"][0]["main"].as<char *>();
                                                                                    ^
In file included from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:9,
                 from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/BasicJsonDocument.hpp:7,
                 from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/DynamicJsonDocument.hpp:7,
                 from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:21,
                 from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from D:\Arduino\M5PaperWeather\weather\Weather.h:25,
                 from D:\Arduino\M5PaperWeather\weather\Data.h:24,
                 from D:\Arduino\M5PaperWeather\weather\weather.ino:30:
D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Object/MemberProxy.hpp:79:7: note: declared here
       as() const {
       ^~
In file included from D:\Arduino\M5PaperWeather\weather\Data.h:24,
                 from D:\Arduino\M5PaperWeather\weather\weather.ino:30:
D:\Arduino\M5PaperWeather\weather\Weather.h:134:84: warning: 'typename ArduinoJson6185_91::enable_if<ArduinoJson6185_91::is_same<T, char*>::value, const char*>::type ArduinoJson6185_91::MemberProxy<TParent, TStringRef>::as() const [with T = char*; TObject = ArduinoJson6185_91::ElementProxy<ArduinoJson6185_91::MemberProxy<ArduinoJson6185_91::ElementProxy<ArduinoJson6185_91::ArrayRef>, const char*> >; TStringRef = const char*; typename ArduinoJson6185_91::enable_if<ArduinoJson6185_91::is_same<T, char*>::value, const char*>::type = const char*]' is deprecated: Replace as<char*>() with as<const char*>() [-Wdeprecated-declarations]
             hourlyIcon[i]    = hourly_list[i - 1]["weather"][0]["icon"].as<char *>();
                                                                                    ^
In file included from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:9,
                 from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/BasicJsonDocument.hpp:7,
                 from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/DynamicJsonDocument.hpp:7,
                 from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:21,
                 from D:\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from D:\Arduino\M5PaperWeather\weather\Weather.h:25,
                 from D:\Arduino\M5PaperWeather\weather\Data.h:24,
                 from D:\Arduino\M5PaperWeather\weather\weather.ino:30:
D:\Arduino\libraries\ArduinoJson\src/ArduinoJson/Object/MemberProxy.hpp:79:7: note: declared here
       as() const {
       ^~
D:\Arduino\libraries\M5EPD\src\M5EPD.cpp: In member function 'void M5EPD::BatteryADCBegin()':
D:\Arduino\libraries\M5EPD\src\M5EPD.cpp:5:25: error: 'ADC1_GPIO35_CHANNEL' was not declared in this scope
 #define BAT_ADC_CHANNEL ADC1_GPIO35_CHANNEL
                         ^~~~~~~~~~~~~~~~~~~
D:\Arduino\libraries\M5EPD\src\M5EPD.cpp:90:31: note: in expansion of macro 'BAT_ADC_CHANNEL'
[... 8 lines cut ...]
D:\Arduino\libraries\M5EPD\src\M5EPD.cpp: In member function 'uint32_t M5EPD::getBatteryRaw()':
D:\Arduino\libraries\M5EPD\src\M5EPD.cpp:5:25: error: 'ADC1_GPIO35_CHANNEL' was not declared in this scope
 #define BAT_ADC_CHANNEL ADC1_GPIO35_CHANNEL
                         ^~~~~~~~~~~~~~~~~~~
D:\Arduino\libraries\M5EPD\src\M5EPD.cpp:100:25: note: in expansion of macro 'BAT_ADC_CHANNEL'
     return adc1_get_raw(BAT_ADC_CHANNEL);
                         ^~~~~~~~~~~~~~~
D:\Arduino\libraries\M5EPD\src\M5EPD.cpp:5:25: note: suggested alternative: 'ADC1_CHANNEL_7'
 #define BAT_ADC_CHANNEL ADC1_GPIO35_CHANNEL
                         ^~~~~~~~~~~~~~~~~~~
D:\Arduino\libraries\M5EPD\src\M5EPD.cpp:100:25: note: in expansion of macro 'BAT_ADC_CHANNEL'
     return adc1_get_raw(BAT_ADC_CHANNEL);
                         ^~~~~~~~~~~~~~~
D:\Arduino\libraries\M5EPD\src\M5EPD.cpp: In member function 'uint32_t M5EPD::getBatteryVoltage()':
[... cut ...]
D:\Arduino\libraries\M5EPD\src\M5EPD.cpp:111:39: note: in expansion of macro 'BAT_ADC_CHANNEL'
         adc_raw_value += adc1_get_raw(BAT_ADC_CHANNEL);
                                       ^~~~~~~~~~~~~~~
D:\Arduino\libraries\M5EPD\src\M5EPD.cpp:5:25: note: suggested alternative: 'ADC1_CHANNEL_7'
 #define BAT_ADC_CHANNEL ADC1_GPIO35_CHANNEL
                         ^~~~~~~~~~~~~~~~~~~
D:\Arduino\libraries\M5EPD\src\M5EPD.cpp:111:39: note: in expansion of macro 'BAT_ADC_CHANNEL'
         adc_raw_value += adc1_get_raw(BAT_ADC_CHANNEL);

guglielmino, 2021-12-03:
This error seems related to some constants names changed in `driver/adc.h`
`M5EPD.cpp` starts defining a constant as below

`#define BAT_ADC_CHANNEL ADC1_GPIO35_CHANNEL`

Looking in `adc.h` it is clear that name of this constant is changed

```
...
 ADC1_CHANNEL_7,     /*!< ADC1 channel 7 is GPIO35 */
...
```

This is confirmed b the compiler output that suggests to use `ADC1_CHANNEL_7` instead of `ADC1_GPIO35_CHANNEL`.

Janaboy, 2022-01-04:
> This error seems related to some constants names changed in `driver/adc.h` `M5EPD.cpp` starts defining a constant as below
> 
> `#define BAT_ADC_CHANNEL ADC1_GPIO35_CHANNEL`
> 
> Looking in `adc.h` it is clear that name of this constant is changed
> 
> ```
> ...
>  ADC1_CHANNEL_7,     /*!< ADC1 channel 7 is GPIO35 */
> ...
> ```
> 
> This is confirmed b the compiler output that suggests to use `ADC1_CHANNEL_7` instead of `ADC1_GPIO35_CHANNEL`.
> 
> Btw, this issue seems fixed looking [here](https://github.com/m5stack/M5EPD/blob/main/src/M5EPD.cpp), but the library linked by the Arduino boards manager JSON file is still broken.

Is there any way to solve this?
Still the same problem :(

GIGObug, 2024-06-08:
I think these errors boil down to installed libraries I'm not sure which ones yet.