Bastelschlumpf/M5PaperWeather #7: Cannot be complied

Janaboy, 2022-01-10:
Dear @Bastelschlumpf , is there any chance to fix the code?
Unfortunately it cannot be compiled, there are multiple issues.

Thank you in advance.

tomschlde, 2022-03-09:
Hi, compiles without errors - there is no need to fix the code.

it is important to add all libraries mentioned in the code (see below) to the Arduino SDK.  Then it will work!
see: https://docs.arduino.cc/software/ide-v1/tutorials/installing-libraries

#include <M5EPD.h>
#include "Config.h"
#include "ConfigOverride.h" // Remove this line
#include "Data.h"
#include "Display.h"
#include "Battery.h"
#include "EPD.h"
#include "EPDWifi.h"
#include "Moon.h"
#include "SHT30.h"
#include "Time.h"
#include "Utils.h"
#include "Weather.h"

Janaboy, 2022-03-09:
> Hi, compiles without errors - there is no need to fix the code.
>
> it is important to add all libraries mentioned in the code (see below) to the Arduino SDK. Then it will work! see: [...]

Can you please help me which libraries version you have installed?
Obviously i have installed all libraries, but give me errors.

Cervidae71, 2022-09-17:
It seems i have the same problem. The software compiles and the readback in serial monitor is all ok. Still it won't display anything. Everytime i compile and upload the code, when done, the screen flashes but still nothing shown.

Serial readout:
M5EPD initializing...OK
Connecting to <<my wifiAP>>
......WiFi connected at: 192.168.0.105
batteryVolt: 4.17
batteryCapacity: 83
Epochtime: 1663452851
DateTime: 17.09.2022 22:14:11
Latitude: 51.0
Longitude: 4.0
WifiRSSI: -50
BatteryVolt: 4.17
BatteryCapacity: 83
Sht30Temperatur: 26
Sht30Humidity: 44
MoonRise: 17.09.2022 20:52:39
MoonSet: 18.09.2022 14:50:36
Sunrise: 17.09.2022 07:19:42
Sunset: 17.09.2022 19:53:50
Winddir: 340.00
Windspeed: 5.14
WeatherDisplay::Show
Stop WiFi
Shutdown

stevejoung, 2023-11-27:
The code fails with the comment "MyData not declared in this scope"

bluejazzCHN, 2023-12-11:
Please notice:
using Arduinojson ver 6.12.0  to compile

orion14794, 2023-12-18:
> > Hi, compiles without errors - there is no need to fix the code.
> 
> > it is important to add all libraries mentioned in the code (see below) to the Arduino SDK. Then it will work! see: https://docs.arduino.cc/software/ide-v1/tutorials/installing-libraries
> > #include <M5EPD.h> #include "Config.h" #include "ConfigOverride.h" // Remove this line #include "Data.h" #include "Display.h" #include "Battery.h" #include "EPD.h" #include "EPDWifi.h" #include "Moon.h" #include "SHT30.h" #include "Time.h" #include "Utils.h" #include "Weather.h"
> 
> Okay an update here. I just installed the m5edp 1.0.3 library and was able to finally compile.

There is no 1.03 M5EPD library file. File versions go up to .01.5 Where did you get your version from???

meldraknailo, 2023-12-29:
I'm pretty sure he meant 0.1.3. You also had a typo in your comment and said .01.5.  =)  I am having issues compling this with ArduinoJson 6.12.0, but ArduinoJson 6.13.0 and M5EPD 0.1.5

meldraknailo, 2023-12-29:
Do you have any idea on how I can get the SHT30 to convert to imperial (F)?

meldraknailo, 2023-12-29:
I was able to get everything to work with USA formatting.  I changed the date to MM:DD:YYYY, the units for the daily to show F by returning imperial units from openweather and by changing the display to show (F) instead of (C).  There are only two things I need help with:  changing the SHT-30 data to show up as F, and the temperature graph doesnt show anything at all for some reason.  The graph draws fine but wont display any data.

GIGObug, 2024-06-10:
Yes please?