Bastelschlumpf/M5PaperWeather #10: Compilation issue

samuelhermann, 2023-01-17:
Hello,

I am really stuck without understanding, I have added Time library https://playground.arduino.cc/Code/Time/ but I am still unable to compile it. Do you have any tips to move as it seems pretty straight forward ?

Thanks

In file included from /home/sam/M5PaperWeather/weather/Weather.h:26,
                 from /home/sam/M5PaperWeather/weather/Data.h:24,
                 from /home/sam/M5PaperWeather/weather/weather.ino:26:
/home/sam/M5PaperWeather/weather/Utils.h:30:10: fatal error: TimeLib.h: No such file or directory
 #include <TimeLib.h>
          ^~~~~~~~~~~
compilation terminated.
Multiple libraries were found for "SD.h"
  Used: /home/sam/.arduino15/packages/m5stack/hardware/esp32/2.0.6/libraries/SD
  Not used: /home/sam/.arduino15/libraries/SD
  Not used: /home/sam/Arduino/libraries/SD-1.2.4
exit status 1

Compilation error: TimeLib.h: No such file or directory

samxplogs, 2023-01-17:
I passed the Time issue on my pc but I am running now with another lib issue

C:\Users\Sam\Documents\GitHub\WeatherM5Paper\Weather\Weather.ino: In function 'void loop()':
C:\Users\Sam\Documents\GitHub\WeatherM5Paper\Weather\Weather.ino:94:33: error: 'class NTPClient' has no member named 'getFormattedDate'; did you mean 'getFormattedTime'?
      formattedDate = timeClient.getFormattedDate();
                                 ^~~~~~~~~~~~~~~~
                                 getFormattedTime
Multiple libraries were found for "SD.h"
  Used: C:\Users\Sam\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.6\libraries\SD
  Not used: C:\Users\Sam\AppData\Local\Arduino15\libraries\SD
exit status 1

Compilation error: 'class NTPClient' has no member named 'getFormattedDate'; did you mean 'getFormattedTime'?

samxplogs, 2023-01-17:
No more ntp issue but I feel I stepped down. I don't know what to check.
This issues are different as I am using legacy arduino on my windows setup instead of the 2nd version on my linux box.

Arduino: 1.8.19 (Windows 10), Board: "M5Stack-Paper, Enabled, Default (2 x 6.5 MB app, 3.6 MB SPIFFS), 240MHz (WiFi/BT), 921600, None, Disabled"

In file included from C:\Users\Sam\Downloads\M5PaperWeather-master\M5PaperWeather-master\weather\Data.h:24,

                 from C:\Users\Sam\Downloads\M5PaperWeather-master\M5PaperWeather-master\weather\weather.ino:26:

C:\Users\Sam\Downloads\M5PaperWeather-master\M5PaperWeather-master\weather\Weather.h: In member function 'bool Weather::Fill(const ArduinoJson6200_F1::JsonObject&)':

Weather.h:117:75: error: no matching function for call to 'ArduinoJson6200_F1::MemberProxy<ArduinoJson6200_F1::ElementProxy<ArduinoJson6200_F1::MemberProxy<ArduinoJson6200_F1::MemberProxy<ArduinoJson6200_F1::JsonObject, const char*>, const char*> >, const char*>::as<char*>()'

       hourlyMain[0]    = root["current"]["weather"][0]["main"].as<char *>();

                                                                           ^

In file included from C:\Users\Sam\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Array/ElementProxy.hpp:7,

                 from C:\Users\Sam\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Array/JsonArray.hpp:7,

                 from C:\Users\Sam\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:24,

                 from C:\Users\Sam\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from C:\Users\Sam\Downloads\M5PaperWeather-master\M5PaperWeather-master\weather\Weather.h:25,

                 from C:\Users\Sam\Downloads\M5PaperWeather-master\M5PaperWeather-master\weather\Data.h:24,

                 from C:\Users\Sam\Downloads\M5PaperWeather-master\M5PaperWeather-master\weather\weather.ino:26:

C:\Users\Sam\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:53:3: note: candidate: 'template<class T> typename ArduinoJson6200_F1::enable_if<(((! ArduinoJson6200_F1::is_same<T, char*>::value) && (! ArduinoJson6200_F1::is_same<T, char>::value)) && (! ArduinoJson6200_F1::ConverterNeedsWriteableRef<T>::value)), T>::type ArduinoJson6200_F1::VariantRefBase<TDerived>::as() const [with T = T; TDerived = ArduinoJson6200_F1::MemberProxy<ArduinoJson6200_F1::ElementProxy<ArduinoJson6200_F1::MemberProxy<ArduinoJson6200_F1::MemberProxy<ArduinoJson6200_F1::JsonObject, const char*>, const char*> >, const char*>]'

   as() const {

   ^~

C:\Users\Sam\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:53:3: note:   template argument deduction/substitution failed:

C:\Users\Sam\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp: In substitution of 'template<class T> typename ArduinoJson6200_F1::enable_if<(((! ArduinoJson6200_F1::is_same<T, char*>::value) && (! ArduinoJson6200_F1::is_same<T, char>::value)) && (! ArduinoJson6200_F1::ConverterNeedsWriteableRef<T>::value)), T>::type ArduinoJson6200_F1::VariantRefBase<ArduinoJson6200_F1::MemberProxy<ArduinoJson6200_F1::ElementProxy<ArduinoJson6200_F1::MemberProxy<ArduinoJson6200_F1::MemberProxy<ArduinoJson6200_F1::JsonObject, const char*>, const char*> >, const char*> >::as<T>() const [with T = char*]':

C:\Users\Sam\Downloads\M5PaperWeather-master\M5PaperWeather-master\weather\Weather.h:117:75:   required from here

C:\Users\Sam\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:53:3: error: no type named 'type' in 'struct ArduinoJson6200_F1::enable_if<false, char*>'

C:\Users\Sam\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:61:3: note: candidate: 'template<class T> typename ArduinoJson6200_F1::enable_if<ArduinoJson6200_F1::ConverterNeedsWriteableRef<T>::value, T>::type ArduinoJson6200_F1::VariantRefBase<TDerived>::as() const [with T = T; TDerived = ArduinoJson6200_F1::MemberProxy<ArduinoJson6200_F1::ElementProxy<ArduinoJson6200_F1::MemberProxy<ArduinoJson6200_F1::MemberProxy<ArduinoJson6200_F1::JsonObject, const char*>, const char*> >, const char*>]'

   as() const {

   ^~

C:\Users\Sam\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:61:3: note:   template argument deduction/substitution failed:

C:\Users\Sam\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp: In substitution of 'template<class T> typename ArduinoJson6200_F1::enable_if<ArduinoJson6200_F1::ConverterNeedsWriteableRef<T>::value, T>::type ArduinoJson6200_F1::VariantRefBase<ArduinoJson6200_F1::MemberProxy<ArduinoJson6200_F1::ElementProxy<ArduinoJson6200_F1::MemberProxy<ArduinoJson6200_F1::MemberProxy<ArduinoJson6200_F1::JsonObject, const char*>, const char*> >, const char*> >::as<T>() const [with T = char*]':

C:\Users\Sam\Downloads\M5PaperWeather-master\M5PaperWeather-master\weather\Weather.h:117:75:   required from here

C:\Users\Sam\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:61:3: error: no type named 'type' in 'struct ArduinoJson6200_F1::enable_if<false, char*>'

In file included from C:\Users\Sam\Downloads\M5PaperWeather-master\M5PaperWeather-master\weather\Data.h:24,

                 from C:\Users\Sam\Downloads\M5PaperWeather-master\M5PaperWeather-master\weather\weather.ino:26:

[... cut ...]
 Used: C:\Users\Sam\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.6\libraries\WiFi

 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi

exit status 1

no matching function for call to 'ArduinoJson6200_F1::MemberProxy<ArduinoJson6200_F1::ElementProxy<ArduinoJson6200_F1::MemberProxy<ArduinoJson6200_F1::MemberProxy<ArduinoJson6200_F1::JsonObject, const char*>, const char*> >, const char*>::as<char*>()'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

closed 2023-01-18 by samuelhermann