Bastelschlumpf/M5PaperWeather #11: MyData not declared in this scope

tgrauss, 2023-05-29:
Hello, when I try to compile, I get the erreor below (MyData not declared in this scope).

---------------------------------------------------
In file included from /home/user/Arduino/M5PaperWeather-master/weather/Utils.h:23,
                 from /home/user/Arduino/M5PaperWeather-master/weather/Weather.h:26,
                 from /home/user/Arduino/M5PaperWeather-master/weather/Data.h:24,
                 from /home/user/Arduino/M5PaperWeather-master/weather/weather.ino:26:
Time.h:26:21: error: 'MyData' was not declared in this scope
 bool SetRTCDateTime(MyData &myData)
                     ^~~~~~
/home/user/Arduino/M5PaperWeather-master/weather/Time.h:26:21: note: suggested alternative: 'FT_Data'
 bool SetRTCDateTime(MyData &myData)
                     ^~~~~~
                     FT_Data
Time.h:26:29: error: 'myData' was not declared in this scope
 bool SetRTCDateTime(MyData &myData)
                             ^~~~~~
/home/user/Arduino/M5PaperWeather-master/weather/Time.h:26:29: note: suggested alternative: 'FT_Data'
 bool SetRTCDateTime(MyData &myData)
                             ^~~~~~
                             FT_Data
/home/user/Arduino/M5PaperWeather-master/weather/weather.ino: In function 'void setup()':
weather:53:31: error: 'SetRTCDateTime' cannot be used as a function
          SetRTCDateTime(myData);
                               ^
In file included from /home/user/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:42,
                 from /home/user/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from /home/user/Arduino/M5PaperWeather-master/weather/Weather.h:25,
                 from /home/user/Arduino/M5PaperWeather-master/weather/Data.h:24,
                 from /home/user/Arduino/M5PaperWeather-master/weather/weather.ino:26:
/home/user/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp: In instantiation of 'static T ArduinoJson::V6212PB::Converter<T, Enable>::fromJson(ArduinoJson::V6212PB::JsonVariantConst) [with T = char*; Enable = void]':
/home/user/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantRefBase.hpp:53:34:   required from 'typename ArduinoJson::V6212PB::detail::enable_if<(! ArduinoJson::V6212PB::detail::ConverterNeedsWriteableRef<T>::value), T>::type ArduinoJson::V6212PB::detail::VariantRefBase<TDerived>::as() const [with T = char*; TDerived = ArduinoJson::V6212PB::detail::MemberProxy<ArduinoJson::V6212PB::detail::ElementProxy<ArduinoJson::V6212PB::detail::MemberProxy<ArduinoJson::V6212PB::detail::MemberProxy<ArduinoJson::V6212PB::JsonObject, const char*>, const char*> >, const char*>; typename ArduinoJson::V6212PB::detail::enable_if<(! ArduinoJson::V6212PB::detail::ConverterNeedsWriteableRef<T>::value), T>::type = char*]'
/home/user/Arduino/M5PaperWeather-master/weather/Weather.h:117:75:   required from here
/home/user/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:24:20: error: no matching function for call to 'convertFromJson(ArduinoJson::V6212PB::JsonVariantConst&, char*&)'
     convertFromJson(src, result);  // Error here? See https://arduinojson.org/v6/unsupported-as/
     ~~~~~~~~~~~~~~~^~~~~~~~~~~~~
/home/user/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:267:13: note: candidate: 'void ArduinoJson::V6212PB::convertFromJson(ArduinoJson::V6212PB::JsonVariantConst, String&)' <near match>
 inline void convertFromJson(JsonVariantConst src, ::String& dst) {
             ^~~~~~~~~~~~~~~
/home/user/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:267:13: note:   conversion of argument 2 would be ill-formed:
/home/user/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:24:26: error: cannot bind non-const lvalue reference of type 'String&' to an rvalue of type 'String'
     convertFromJson(src, result);  // Error here? See https://arduinojson.org/v6/unsupported-as/
                          ^~~~~~
In file included from /home/user/.arduino15/packages/m5stack/hardware/esp32/2.0.7/cores/esp32/Arduino.h:166,
                 from sketch/weather.ino.cpp:1:
/home/user/.arduino15/packages/m5stack/hardware/esp32/2.0.7/cores/esp32/WString.h:57:9: note:   after user-defined conversion: 'String::String(const char*)'
         String(const char *cstr = "");
         ^~~~~~
In file included from /home/user/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:42,
                 from /home/user/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from /home/user/Arduino/M5PaperWeather-master/weather/Weather.h:25,
                 from /home/user/Arduino/M5PaperWeather-master/weather/Data.h:24,
                 from /home/user/Arduino/M5PaperWeather-master/weather/weather.ino:26:
/home/user/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:283:13: note: candidate: 'void ArduinoJson::V6212PB::convertFromJson(ArduinoJson::V6212PB::JsonVariantConst, std::__cxx11::string&)' <near match>
 inline void convertFromJson(JsonVariantConst src, std::string& dst) {
             ^~~~~~~~~~~~~~~
/home/user/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:283:13: note:   conversion of argument 2 would be ill-formed:
/home/user/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:24:26: error: cannot bind non-const lvalue reference of type 'std::__cxx11::string&' {aka 'std::__cxx11::basic_string<char>&'} to an rvalue of type 'std::__cxx11::string' {aka 'std::__cxx11::basic_string<char>'}
     convertFromJson(src, result);  // Error here? See https://arduinojson.org/v6/unsupported-as/
                          ^~~~~~
In file included from /home/user/.arduino15/packages/m5stack/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/string:52,
                 from /home/user/.arduino15/packages/m5stack/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/stdexcept:39,
                 from /home/user/.arduino15/packages/m5stack/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/array:39,
                 from /home/user/.arduino15/packages/m5stack/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/tuple:39,
                 from /home/user/.arduino15/packages/m5stack/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/functional:54,
                 from /home/user/.arduino15/packages/m5stack/hardware/esp32/2.0.7/cores/esp32/HardwareSerial.h:49,
                 from /home/user/.arduino15/packages/m5stack/hardware/esp32/2.0.7/cores/esp32/Arduino.h:174,
                 from sketch/weather.ino.cpp:1:
/home/user/.arduino15/packages/m5stack/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/basic_string.h:514:7: note:   after user-defined conversion: 'std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::basic_string(const _CharT*, const _Alloc&) [with _CharT = char; _Traits = std::char_traits<char>; _Alloc = std::allocator<char>]'
       basic_string(const _CharT* __s, const _Alloc& __a = _Alloc())
       ^~~~~~~~~~~~
Multiple libraries were found for "WiFiClient.h"
 Used: /home/user/.arduino15/packages/m5stack/hardware/esp32/2.0.7/libraries/WiFi
 Not used: /home/user/arduino-1.8.15/libraries/WiFi
Multiple libraries were found for "SD.h"
 Used: /home/user/.arduino15/packages/m5stack/hardware/esp32/2.0.7/libraries/SD
 Not used: /home/user/arduino-1.8.15/libraries/SD
exit status 1
'MyData' was not declared in this scope

stevejoung, 2023-11-27:
Same issue!

ZeFish, 2024-08-16:
Same here, ... don't know what do to since it seems to works for other.