Bastelschlumpf/M5PaperWeather #12: Can't burn sketch to my M5Paper V1.1

orion14794, 2023-12-18:
I have installed all libraries and all required to get my M5Paper detected and loaded on Arduino. I am able to upload other sketches to my device, except this one. Here is the error message I get: 

Arduino: 1.8.19 (Windows 10), Board: "M5Paper, Enabled, Default (2 x 6.5 MB app, 3.6 MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 16MB (128Mb), 1500000, Core 1, Core 1, None, Disabled"

\\weather.ino.cpp" -o "C:\\Users\\johnn\\AppData\\Local\\Temp\\arduino_build_670619\\sketch\\weather.ino.cpp.o"

In file included from C:\Users\johnn\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:42,

                 from C:\Users\johnn\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from D:\Downloads\TTGOwatch+M5Stack stuff\M5Paper stuff\M5PaperWeather\M5PaperWeather\weather\Weather.h:25,

                 from D:\Downloads\TTGOwatch+M5Stack stuff\M5Paper stuff\M5PaperWeather\M5PaperWeather\weather\Data.h:24,

                 from D:\Downloads\TTGOwatch+M5Stack stuff\M5Paper stuff\M5PaperWeather\M5PaperWeather\weather\weather.ino:26:

C:\Users\johnn\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp: In instantiation of 'static T ArduinoJson::V6214PB2::Converter<T, Enable>::fromJson(ArduinoJson::V6214PB2::JsonVariantConst) [with T = char*; Enable = void]':

C:\Users\johnn\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:53:34:   required from 'typename ArduinoJson::V6214PB2::detail::enable_if<(! ArduinoJson::V6214PB2::detail::ConverterNeedsWriteableRef<T>::value), T>::type ArduinoJson::V6214PB2::detail::VariantRefBase<TDerived>::as() const [with T = char*; TDerived = ArduinoJson::V6214PB2::detail::MemberProxy<ArduinoJson::V6214PB2::detail::ElementProxy<ArduinoJson::V6214PB2::detail::MemberProxy<ArduinoJson::V6214PB2::detail::MemberProxy<ArduinoJson::V6214PB2::JsonObject, const char*>, const char*> >, const char*>; typename ArduinoJson::V6214PB2::detail::enable_if<(! ArduinoJson::V6214PB2::detail::ConverterNeedsWriteableRef<T>::value), T>::type = char*]'

D:\Downloads\TTGOwatch+M5Stack stuff\M5Paper stuff\M5PaperWeather\M5PaperWeather\weather\Weather.h:115:75:   required from here

C:\Users\johnn\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp:32:20: error: no matching function for call to 'convertFromJson(ArduinoJson::V6214PB2::JsonVariantConst&, char*&)'

     convertFromJson(src, result);  // Error here? See https://arduinojson.org/v6/unsupported-as/

     ~~~~~~~~~~~~~~~^~~~~~~~~~~~~

C:\Users\johnn\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp:275:13: note: candidate: 'void ArduinoJson::V6214PB2::convertFromJson(ArduinoJson::V6214PB2::JsonVariantConst, String&)' <near match>

 inline void convertFromJson(JsonVariantConst src, ::String& dst) {

             ^~~~~~~~~~~~~~~

C:\Users\johnn\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp:275:13: note:   conversion of argument 2 would be ill-formed:

C:\Users\johnn\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp:32:26: error: cannot bind non-const lvalue reference of type 'String&' to an rvalue of type 'String'

     convertFromJson(src, result);  // Error here? See https://arduinojson.org/v6/unsupported-as/

                          ^~~~~~

In file included from C:\Users\johnn\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.9\cores\esp32/Arduino.h:176,

                 from C:\Users\johnn\AppData\Local\Temp\arduino_build_670619\sketch\weather.ino.cpp:1:

C:\Users\johnn\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.9\cores\esp32/WString.h:59:9: note:   after user-defined conversion: 'String::String(const char*)'

         String(const char *cstr = "");

         ^~~~~~

In file included from C:\Users\johnn\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:42,

                 from C:\Users\johnn\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from D:\Downloads\TTGOwatch+M5Stack stuff\M5Paper stuff\M5PaperWeather\M5PaperWeather\weather\Weather.h:25,

                 from D:\Downloads\TTGOwatch+M5Stack stuff\M5Paper stuff\M5PaperWeather\M5PaperWeather\weather\Data.h:24,

                 from D:\Downloads\TTGOwatch+M5Stack stuff\M5Paper stuff\M5PaperWeather\M5PaperWeather\weather\weather.ino:26:

C:\Users\johnn\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp:291:13: note: candidate: 'void ArduinoJson::V6214PB2::convertFromJson(ArduinoJson::V6214PB2::JsonVariantConst, std::__cxx11::string&)' <near match>

 inline void convertFromJson(JsonVariantConst src, std::string& dst) {

             ^~~~~~~~~~~~~~~

C:\Users\johnn\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp:291:13: note:   conversion of argument 2 would be ill-formed:

[... 46 lines cut ...]
Using library Wire at version 2.0.0 in folder: C:\Users\johnn\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.9\libraries\Wire 

Using library SPI at version 2.0.0 in folder: C:\Users\johnn\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.9\libraries\SPI 

Using library FS at version 2.0.0 in folder: C:\Users\johnn\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.9\libraries\FS 

Using library SD at version 2.0.0 in folder: C:\Users\johnn\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.9\libraries\SD 

Using library SPIFFS at version 2.0.0 in folder: C:\Users\johnn\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.9\libraries\SPIFFS 

Using library HTTPClient at version 2.0.0 in folder: C:\Users\johnn\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.9\libraries\HTTPClient 

Using library WiFi at version 2.0.0 in folder: C:\Users\johnn\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.9\libraries\WiFi 

Using library WiFiClientSecure at version 2.0.0 in folder: C:\Users\johnn\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.9\libraries\WiFiClientSecure 

Using library ArduinoJson at version 6.21.4 in folder: C:\Users\johnn\Documents\Arduino\libraries\ArduinoJson 

Using library Time at version 1.6.1 in folder: C:\Users\johnn\Documents\Arduino\libraries\Time 

Using library MoonRise at version 2.0.4 in folder: C:\Users\johnn\Documents\Arduino\libraries\MoonRise 

exit status 1

Error compiling for board M5Paper.

nicolasborn, 2023-12-27:
I had the same issue. Just use ArduinoJson version 6.12.0 to compile.

jdbaluch, 2023-12-28:
I am able to compile using version 6.12.0 and the software loads up on the M5Paper v1.1; however when reviewing the serial output I get` GetWeather failed, error: DateTime: 26.09.2020 16:56:53 `and the device is never able to connect.  I'm connected successfully to my wifi.  It's as though the JSON isn't being parsed.  Any guidance?  Thanks!

nicolasborn, 2023-12-28:
Hi! You need a valid subscription for the oneCall API. The first 1000 calls per day are free but you need to register a credit card anyway. And I changed the uri in Weather.h to version 3.0: uri += "/data/3.0/onecall“;

meldraknailo, 2023-12-29:
I tried using ArduinoJson 6.12.0 to complie and I am still having issues with compiling the sketch.  On a side note, you can also set a limit to the calls per day to ensure you dont get charged.  I use the 3.0 call service for several different weather apps on M5Paper and M5Stack devices and dont get close to the 1,000 calls per day limit.

Here are my error messages if anyone can provide guidance:
In file included from /Users/X/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Array/ArrayIterator.hpp:8,
                 from /Users/X/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Array/ArrayRef.hpp:8,
                 from /Users/X/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:17,
                 from /Users/X/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from /Users/X/Downloads/M5PaperWeather-master/weather/Weather.h:25,
                 from /Users/X/Downloads/M5PaperWeather-master/weather/Data.h:24,
                 from /Users/X/Downloads/M5PaperWeather-master/weather/weather.ino:25:
/Users/X/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantRef.hpp:12:10: fatal error: ArduinoJson/Operators/VariantOperators.hpp: No such file or directory
 #include <ArduinoJson/Operators/VariantOperators.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1

meldraknailo, 2023-12-29:
Okay so I was able to get it to compile several times using different combinations of ArduinoJson and M5EPD versions, but every time I upload it I get a blank screen.
I can load a repository based on this one fine: https://github.com/marcotammaro/M5Paper---Modern-Weather-Dashboard

GIGObug, 2024-06-09:
> I figured it out. ArduinoJson 6.12.0 and M5EPD 0..1.5 worked and it now runs great!

WOW!!!
Thank you!
It compiled, and uploaded. Now I need to change the data to Imperial.