Bastelschlumpf/M5PaperWeather #16: Compilation error: 'rtc_date_t' was not declared in this scope

IR4TE, 2024-12-18:
I admit I'm new to this stuff, I followed what others already said which libraries should work but always end with this error when trying to compile.

In file included from C:\Users\irate\Desktop\M5PaperWeather-master\weather\Weather.h:26,
                 from C:\Users\irate\Desktop\M5PaperWeather-master\weather\Data.h:24,
                 from C:\Users\irate\Desktop\M5PaperWeather-master\weather\weather.ino:24:
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Utils.h: In function 'String getRTCDateTimeString()':
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Utils.h:30:4: error: 'rtc_date_t' was not declared in this scope
    rtc_date_t date_struct;
    ^~~~~~~~~~
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Utils.h:30:4: note: suggested alternative: 'rmt_data_t'
    rtc_date_t date_struct;
    ^~~~~~~~~~
    rmt_data_t
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Utils.h:31:4: error: 'rtc_time_t' was not declared in this scope
    rtc_time_t time_struct;
    ^~~~~~~~~~
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Utils.h:31:4: note: suggested alternative: 'dhcps_time_t'
    rtc_time_t time_struct;
    ^~~~~~~~~~
    dhcps_time_t
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Utils.h:33:4: error: 'M5' was not declared in this scope
    M5.RTC.getDate(&date_struct);
    ^~
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Utils.h:33:4: note: suggested alternative: 'G5'
    M5.RTC.getDate(&date_struct);
    ^~
    G5
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Utils.h:33:20: error: 'date_struct' was not declared in this scope
    M5.RTC.getDate(&date_struct);
                    ^~~~~~~~~~~
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Utils.h:33:20: note: suggested alternative: 'uart_struct_t'
    M5.RTC.getDate(&date_struct);
                    ^~~~~~~~~~~
                    uart_struct_t
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Utils.h:34:20: error: 'time_struct' was not declared in this scope
    M5.RTC.getTime(&time_struct);
                    ^~~~~~~~~~~
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Utils.h:34:20: note: suggested alternative: 'timeNotSet'
    M5.RTC.getTime(&time_struct);
                    ^~~~~~~~~~~
                    timeNotSet
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Utils.h: In function 'time_t GetRTCTime()':
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Utils.h:47:3: error: 'rtc_date_t' was not declared in this scope
   rtc_date_t   date_struct;
   ^~~~~~~~~~
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Utils.h:47:3: note: suggested alternative: 'rmt_data_t'
   rtc_date_t   date_struct;
   ^~~~~~~~~~
   rmt_data_t
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Utils.h:48:3: error: 'rtc_time_t' was not declared in this scope
   rtc_time_t   time_struct;
   ^~~~~~~~~~
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Utils.h:48:3: note: suggested alternative: 'dhcps_time_t'
   rtc_time_t   time_struct;
   ^~~~~~~~~~
   dhcps_time_t
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Utils.h:50:3: error: 'M5' was not declared in this scope
   M5.RTC.getDate(&date_struct);
   ^~
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Utils.h:50:3: note: suggested alternative: 'G5'
   M5.RTC.getDate(&date_struct);
   ^~
   G5
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Utils.h:50:19: error: 'date_struct' was not declared in this scope
   M5.RTC.getDate(&date_struct);
                   ^~~~~~~~~~~
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Utils.h:50:19: note: suggested alternative: 'uart_struct_t'
   M5.RTC.getDate(&date_struct);
                   ^~~~~~~~~~~
                   uart_struct_t
[... 340 lines cut ...]
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Time.h:31:7: note: suggested alternative: 'rmt_data_t'
       rtc_date_t RTCDate;
       ^~~~~~~~~~
       rmt_data_t
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Time.h:35:7: error: 'RTCDate' was not declared in this scope
       RTCDate.year = year(time);
       ^~~~~~~
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Time.h:38:7: error: 'M5' was not declared in this scope
       M5.RTC.setDate(&RTCDate);
       ^~
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Time.h:38:7: note: suggested alternative: 'G5'
       M5.RTC.setDate(&RTCDate);
       ^~
       G5
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Time.h:40:7: error: 'RTCtime' was not declared in this scope
       RTCtime.hour = hour(time);
       ^~~~~~~
C:\Users\irate\Desktop\M5PaperWeather-master\weather\Time.h:40:7: note: suggested alternative: 'time'
       RTCtime.hour = hour(time);
       ^~~~~~~
       time

exit status 1

Compilation error: 'rtc_date_t' was not declared in this scope

Bastelschlumpf, 2024-12-19:
This definition is part of the M5Paper library
In my case in the folder
c:\Users\xxx\Documents\Arduino\libraries\M5EPD\src\utility\BM8563.h
Either you have not yet installed the whole M5Stack library via the Board Manager or you have not yet selected the M5Paper board in your Arduino environment.
Tools/Boards/M5Stack Arduino/M5StackPaper

IR4TE, 2024-12-20:
Everything is installed and the M5Paper is selected in the whole list of M5 Boards

Bastelschlumpf, 2024-12-21:
The definition is in my case in the file
c:\Users\[user]\Documents\Arduino\libraries\M5EPD\src\utility\BM8563.h
I used Arduino IDE 2.3.1 with the
M5Stack Arduino Library 1.0.7
No idea, what the problem could be.
Search the BM8563.h file and check if the "rtc_date_t" definition is in the file.

designdevy, 2024-12-30:
Same issue as well.
Compilation error: 'rtc_date_t' was not declared in this scope

designdevy, 2024-12-31:
Ok I was able to fix it by deleting arduino ide completely and installing IDE 2.3.1 m5stack board and libraries from scratch.